Pinoccio/library-pinoccio #236: Can't call ScoutScript commands (or the functions they call) in Shell/Arduino IDE

cylinderStudio, 2015-07-21:
Using the Bootstrap.ino, for example, I can't make calls to ScoutScript commands, or to the functions they call, from within an Arduino sketch. 

```
temperature.report()
```
Seems to work, but I've tried the following to no avail:

```
pin.setmode('d7', INPUT_PULLUP);
// 'pin' was not declared in this scope

pinSetMode('d7', INPUT_PULLUP);
// 'pinSetMode' was not declared in this scope

Shell.eval('pin.setmode','d7',INPUT_PULLUP);
// no matching function for call to 'PinoccioShell::eval(int, int, int)'

matthijskooijman, 2015-07-23:
You're using single quotes, which normally contain only a single character in C/C++. You should be using double quotes for strings instead. Let me know if that helps.

matthijskooijman, 2015-07-23:
Also, when calling pin.setmode, IIRC you need to use the `PINMODE_` constants (since setmode supports more modes than the Arduino builtin `pinMode`, which is what `INPUT_PULLUP` is for). So:

    Shell.eval("pin.setmode", "d7", PINMODE_INPUT_PULLUP);

Alternatively, call:

    Scout.setMode(7, PINMODE_INPUT_PULLUP)

This needs an integer pin number instead of a pin name.

cylinderStudio, 2015-07-23:
Exactly what I needed to know. I discovered the error with the single quote after a bit of debugging, but was lost otherwise. Thanks.

closed 2015-07-23 by cylinderStudio